RapidScada/scada #8: turkish chracter issue

epoch79, 2015-07-23:
hi, I want to report a minor bug. 
When you click pass to server button on Admin application, the app creating "ınterface.dat" not the "interface.dat" in the C:\SCADA\BaseDAT directory. I think this is a lower case issue in Admin application on  "Tables" class's file name property.
public string FileName
            {
                get
                {
                    return Name.ToLower() + ".dat";
                }
            }

2mik, 2015-07-23:
Thanks! Could you give additional information:
1. Is the symbol 'ı' a letter of Turkish alphabet?
2. How this issue affects work of the software?

2mik, 2015-07-23:
I have fixed this issue in develop branch. Tell me, please, if it affects normal work, and I'll merge it to the distributive.

epoch79, 2015-07-24:
yes "ı" is letter.
when you click pass to server button in basedat folder "ınterface.dat"  and "ıncnl.dat".
Changes on Admin app doesn't apply to server.

epoch79, 2015-07-24:
i tried your fix and working good. thanks.
But i think you have to merge it to the distributive immediately.

2mik, 2015-07-27:
Done.

closed 2015-07-27 by 2mik