RapidScada/scada #15: ScadaAdmin.exe crash in win10

codersmurf, 2015-10-01:
When running the application it never seems to start and creates a dump.  I have some but git doesn't seem to support attaching them, here's the info to one though:

Dump Summary
------------
Dump File:	ScadaAdmin.exe.6884.dmp : C:\Users\Jacob\AppData\Local\CrashDumps\ScadaAdmin.exe.6884.dmp
Last Write Time:	9/30/2015 3:25:07 PM
Process Name:	ScadaAdmin.exe : C:\SCADA\ScadaAdmin\ScadaAdmin.exe
Process Architecture:	x86
Exception Code:	0xE0434352
Exception Information:	
Heap Information:	Not Present

System Information
------------------
OS Version:	10.0.10240
CLR Version(s):	4.6.96.0

Modules
-------
Module Name	Module Path	Module Version
-----------	-----------	--------------
ScadaAdmin.exe	C:\SCADA\ScadaAdmin\ScadaAdmin.exe	4.0.0.0
ntdll.dll	C:\Windows\System32\ntdll.dll	10.0.10240.16430
mscoree.dll	C:\Windows\System32\mscoree.dll	10.0.10240.16384
kernel32.dll	C:\Windows\System32\kernel32.dll	10.0.10240.16384
KERNELBASE.dll	C:\Windows\System32\KERNELBASE.dll	10.0.10240.16384
apphelp.dll	C:\Windows\System32\apphelp.dll	10.0.10240.16384
advapi32.dll	C:\Windows\System32\advapi32.dll	10.0.10240.16384
msvcrt.dll	C:\Windows\System32\msvcrt.dll	7.0.10240.16384
sechost.dll	C:\Windows\System32\sechost.dll	10.0.10240.16384
rpcrt4.dll	C:\Windows\System32\rpcrt4.dll	10.0.10240.16412
sspicli.dll	C:\Windows\System32\sspicli.dll	10.0.10240.16384
CRYPTBASE.dll	C:\Windows\System32\CRYPTBASE.dll	10.0.10240.16384
bcryptPrimitives.dll	C:\Windows\System32\bcryptPrimitives.dll	10.0.10240.16384
mscoreei.dll	C:\Windows\Microsoft.NET\Framework\v4.0.30319\mscoreei.dll	4.6.79.0
shlwapi.dll	C:\Windows\System32\shlwapi.dll	10.0.10240.16384
combase.dll	C:\Windows\System32\combase.dll	10.0.10240.16384
gdi32.dll	C:\Windows\System32\gdi32.dll	10.0.10240.16390
user32.dll	C:\Windows\System32\user32.dll	10.0.10240.16384
imm32.dll	C:\Windows\System32\imm32.dll	10.0.10240.16384
msctf.dll	C:\Windows\System32\msctf.dll	10.0.10240.16384
kernel.appcore.dll	C:\Windows\System32\kernel.appcore.dll	10.0.10240.16384
version.dll	C:\Windows\System32\version.dll	10.0.10240.16384
clr.dll	C:\Windows\Microsoft.NET\Framework\v4.0.30319\clr.dll	4.6.96.0
MSVCR120_CLR0400.dll	C:\Windows\System32\MSVCR120_CLR0400.dll	12.0.52512.0
mscorlib.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\mscorlib\739821138ef3780d29566f16c4ec3836\mscorlib.ni.dll	4.6.96.0
ole32.dll	C:\Windows\System32\ole32.dll	10.0.10240.16384
uxtheme.dll	C:\Windows\System32\uxtheme.dll	10.0.10240.16397
tiptsf.dll	C:\Program Files (x86)\Common Files\Microsoft Shared\Ink\tiptsf.dll	10.0.10240.16384
oleaut32.dll	C:\Windows\System32\oleaut32.dll	10.0.10240.16384
clrjit.dll	C:\Windows\Microsoft.NET\Framework\v4.0.30319\clrjit.dll	4.6.96.0
System.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\System\ca6efbdfe42568e069a7e2f1a869de59\System.ni.dll	4.6.79.0
System.Drawing.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\System.Drawing\ed0563bc783d7cf316e280e8d17770eb\System.Drawing.ni.dll	4.6.93.0
System.Windows.Forms.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\System.Windows.Forms\eabdc05eaf0b37bf4dc02b4a64b1d491\System.Windows.Forms.ni.dll	4.6.79.0
comctl32.dll	C:\Windows\WinSxS\x86_microsoft.windows.common-controls_6595b64144ccf1df_5.82.10240.16384_none_49c02355cf03478c\comctl32.dll	5.82.10240.16384
comctl32.dll	C:\Windows\WinSxS\x86_microsoft.windows.common-controls_6595b64144ccf1df_6.0.10240.16384_none_3bccb1ff6bcd1849\comctl32.dll	6.10.10240.16384
dwmapi.dll	C:\Windows\System32\dwmapi.dll	10.0.10240.16392
cryptsp.dll	C:\Windows\System32\cryptsp.dll	10.0.10240.16384
bcrypt.dll	C:\Windows\System32\bcrypt.dll	10.0.10240.16384
rsaenh.dll	C:\Windows\System32\rsaenh.dll	10.0.10240.16384
System.Core.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\System.Core\c443a8daf03f55a14ffe24ec583664da\System.Core.ni.dll	4.6.79.0
System.Configuration.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\System.Configuration\f57c97ac6e9e80661adc3b51be05a50b\System.Configuration.ni.dll	4.6.79.0
System.Xml.ni.dll	C:\Windows\assembly\NativeImages_v4.0.30319_32\System.Xml\e6dca958930c9fab108edde22f45f2d0\System.Xml.ni.dll	4.6.79.0
GdiPlus.dll	C:\Windows\WinSxS\x86_microsoft.windows.gdiplus_6595b64144ccf1df_1.1.10240.16384_none_d15682eeaf714889\GdiPlus.dll	10.0.10240.16384
DWrite.dll	C:\Windows\System32\DWrite.dll	10.0.10240.16430
WindowsCodecs.dll	C:\Windows\System32\WindowsCodecs.dll	10.0.10240.16384
diasymreader.dll	C:\Windows\Microsoft.NET\Framework\v4.0.30319\diasymreader.dll	14.0.79.0

2mik, 2015-10-01:
I think, MSSQL Compact 3.5 was not installed. There should be information in setup log. Could you reinstall SCADA-Admin and post the log here?

2mik, 2015-10-01:
Rapid SCADA works correctly on Windows 10, but could be some nuances.

2mik, 2015-10-01:
.NET Framework 3.5 is required for SCADA-Admin. Please check whether it is installed. In the future this dependency will be removed.

closed 2015-10-14 by 2mik