RapidScada/scada #17: SIEMENS modbus card direct connection support

zealwebtech, 2015-10-14:
Hi,
Is that we can replace ADAM cponverter and can connect directly with SIEMENS modbus cards and devices for pooling the data?

2mik, 2015-10-14:
I think yes )

zealwebtech, 2015-10-21:
Hi Mik,
I was testing it along with SIEMENS S7-415 PLC using SIEMENS PCS7 with Cp-341 over RS485 i was facing error like External error occured.
Can you have some manual or procedure?
I really wanted to impliment this with SIEMENS products.

2mik, 2015-10-21:
Hi,
Download the latest Rapid SCADA installation package from http://rapidscada.org/
In Doc folder there is a step by step manual describes how to set up Mobdus on Rapid SCADA.
Please provide the information about the error shown in SCADA-Communicator communication line log.

closed 2015-12-15 by 2mik