RapidScada/scada #23: Graphical designer for RapidScada

mcgiwer, 2016-07-11:
Please create a graphical editor for RapidScada with would work on Windows and Linux systems, with would allow to create own "interfaces" instead of having only the demo ones

2mik, 2016-07-11:
Web editor is in the plans 2016.

mcgiwer, 2016-07-11:
@2mik Any idea when the works could start as soonest?

2mik, 2016-07-11:
This is a major task. Do you have a deadline?
The schemes and tables editors will be developed as a web apps, but SCADA-Administrator and Server's and Communicator's shells still remain Windows desktop applications.

mcgiwer, 2016-07-11:
I don't have a deadline yet but I would like to earlier design and plan the whole instalation and it's controls before starting to implement it

2mik, 2016-07-11:
I'd like not to promise an exact release date because it may be changed due to work load. The date can be fixed in case of commercial contract. Anyway you can use the current Windows GUI version of the editor while the new version is under construction.

mcgiwer, 2016-07-11:
> Anyway you can use the current Windows GUI version of the editor while the new version is under construction.

The problem is that I use Linux (Ubuntu) and there is not sure if it will run on iy

2mik, 2016-07-12:
Porting all the apps to web requires huge LoE. As far as I know, you can use Windows emulator on Linux. Probably it is possible to launch GUI using Mono, but it requires a research. You can try to do it by yourself or we can discuss custom development if it is possible to pay for work.

closed 2016-08-14 by 2mik