RapidScada/scada #24: Impossible to create SNMP variable

DjMomo, 2016-09-15:
Hello,

I'm a new user of Rapid Scada. It's seems to be a great product.

I've got a problem with SNMP Module. I'm trying to retrieve SNMP values, but I can't create a variable in _SCADA Communicator -> My Line -> My Device configuration -> Add variable_.
In _Owner name_ field I add a variable Name. 
In _OID_ field I add requested OID (for example) : `.1.3.6.4.1.6741.1.2.0`
When I clic Add button, _Incorrect OID_ Error message is displayed and variable isn't created.

This message is generated by `private bool CheckOID(string oidStr)` function (ScadaComm/OpenKPs/KpSnmp/FrmVariable.cs).

Do you know why ?

Is it an OID string format problem ?

Can you send me (or publish) a SNMP & OID configuration example ?

Thanks

DjMomo, 2016-09-15:
OK,  problem is due to my server (2003 Server). Works fine on Seven.

2mik, 2016-09-15:
Hello,
Thanks for your the info that the problem is solved.

closed 2016-09-15 by 2mik