RapidScada/scada #30: Projects Looks Interesting, is it possible to develop its view on Ios and Android

abhishekricky, 2016-12-25:
Hi,

I was looking for possibilities, to have application also for ios and android, becoz tht would complete its cycle

2mik, 2016-12-26:
Hi,
Of course, it is possible to develop iOS and Android native applications for Rapid SCADA. Rapid SCADA has rich integration options: TCP, HTTP, DB. But such tasks currently not in our plans. If you decide develop these apps, I will provide necessary information.

abhishekricky, 2016-12-26:
Hi,

Yes. I am planning to start it development will take Android application
first. Will create github repo .

About integration in 2 days, I will create roadmap and additional
information that is required

2mik, 2016-12-28:
Please ask in this forum http://forum.rapidscada.org/?forum=development-and-integration

closed 2016-12-28 by 2mik