RapidScada/scada #35: Not able build using VS-2013

subu1979, 2017-06-06:
Hi Team,

As per the HowToBuild.txt file, i'm not able to build the project. Can you provide us the base solution(sln) file for the whole project. Also where is the following projects present.

1) OpenKPs
2) ScadaCommonCommon

Request you to kindly do the needful.

Regards
Subramaniyam.KMV

2mik, 2017-06-07:
Hello,
What error messages did you receive during build?
Joining a lot of projects into a one solution significantly reduce the speed of Visual Studio,
[OpenKPs](https://github.com/RapidScada/scada/blob/master/ScadaComm/OpenKPs/OpenKPs.sln)
ScadaCommCommon is a part of [ScadaComm](https://github.com/RapidScada/scada/blob/master/ScadaComm/ScadaComm.sln)

subu1979, 2017-06-07:
Hi,

Thanks for your information. I was able to build the project successfully.

Regards
Subramaniyam.KMV

closed 2017-06-08 by 2mik