RapidScada/scada #38: ScadaWebCommon-UserData.cs : ServerComm.Roles is undefined

tamtrong, 2017-07-12:
Hi Everyone,
I am newbie to RapidScada, after downloading sourcecode and build the solution in order of HowToBuild.txt, when building ScadaWebCommon project which ref to ScadaData project, I found this error:
Error	CS0426	The type name 'Roles' does not exist in the type 'ServerComm'
Therefore, I checked ServerComm.cs of ScadaData project and found no property named "Roles"
I've download the latest version sourcecode until now.
Could somebody explain this? Many thanks

2mik, 2017-07-12:
Hi,
Some of the projects, including ScadaWebCommon,  are obsolete. We will clean then in the near future.
Build ScadaWebCommon5 and ScadaWebShell5. Skip: ScadaWebCommon, ScadaWebConfig, WebPageSample.

2mik, 2017-07-12:
Also skip ScadaWeb\OpenPlugins\PlgScheme

tamtrong, 2017-07-12:
Thanks for your information, 2mik. RapidScada is really interesting to me.

2mik, 2017-07-12:
Thank you too ))

closed 2017-07-12 by 2mik