RapidScada/scada #46: Porting to .net standard

circler3, 2018-02-01:
### Ideas

Hi,
Since framework 4.0 lacks some key feature (such as async/await) and .net standard is becoming a popluar choice for developers, how about changing the target framework to .net standard 2.0? 

> Port to .net framework 4.6.1 for winforms and .net standard 2.0 for libraries. Because .net framework  4.6.1 is shipped with win10 and in compliance with .net standard 2.0

I am willing to help handle the porting process.

2mik, 2018-02-01:
Hi,
It's planned to move to .NET 4.6 or 4.7 when we get rid of using MSSQL compact. But I can't provide the exact date.

What are the benefits of changing libraries target to .NET standard 2.0, by your opinion?

2mik, 2018-02-01:
Have you tested .NET standard libraries with Mono? See https://stackoverflow.com/questions/48061333/using-net-standard-2-0-with-mono

circler3, 2018-02-02:
Hi,
Targeting .NET standard will eliminate compatibility issues when library is used in other platform or .NET implementations (such as .net core). If I use Mono 5.4 or later versions, I do not need to do library test since Mono 5.4 support .net standard 2.0

[This page](https://docs.microsoft.com/en-us/dotnet/standard/components) says It(Mono) supports all of the currently published .NET Standard versions.

[This page](https://docs.microsoft.com/en-us/dotnet/standard/net-standard) indicates that Mono 5.4 support .net standard 2.0. So I don't think it will be an issue when targeting mono.

P.S. I think standard libraries will be widely used in the future.
But I have several concerns:
1. Some third-party libraries may be obsacles when porting net standard. E.g. mysql connector library who support net standard lacked `Adapter` Classs several months ago (I have no idea whether it is ready now or not).
2. .NET standard binds *nuget*. Maybe the program would be "messed up" with a lot of dlls.

circler3, 2018-02-02:
Hi,
Since you are planing to port to a newer framework and going to get rid of the mssqlcompact, how about listing a roadmap about the incoming features without certain deadlines at your convenience? I think README.MD or opening a new issue or wikis are good choices. It can be more attractive for other developers to participate in the project.

2mik, 2018-02-03:
Hi,
I hope, Microsoft will not close their idea of support cross-platform .NET standard. This approach seems proper.
The preferable of communicating is http://forum.rapidscada.org/ Are you registered? If you create a new topic "Roadmap", I will post the roadmap there with pleasure.

I will close this GitHub issue, I mean it is On Hold, but I don't see no such state here.

closed 2018-02-03 by 2mik