RapidScada/scada #52: Basic Components Not Visible in Scheme Editor or Webstation

DmanDman, 2018-04-04:
Basic components are not visible in Scheme Editor. Webstation also does not display basic components. I have installed PlgSchBasicComp.dll on Webstation, but the basic components in my scheme files are not being rendered. Please advise.

2mik, 2018-04-06:
Hi,
Please provide version of the Webstation app and Scheme Editor.
Also copy here errors that may be in C:\SCADA\ScadaWeb\log\ScadaWeb.log and C:\SCADA\ScadaSchemeEditor\log\ScadaSchemeEditor.log

>  I have installed PlgSchBasicComp.dll on Webstation
They should be installed automatically after the web app installation. May be you should download actual version of Rapid SCADA.

DmanDman, 2018-04-09:
**Webstation** 1.0.0
**Scheme Editor** 5.2.1.0

**Log File**:

2018-04-09 10:55:20 <DESKTOP-99VORTJ><DefaultAppPool><ACT> Initialize common web application data
2018-04-09 10:55:21 <DESKTOP-99VORTJ><DefaultAppPool><EXC> Error initializing plugin "Basic scheme components":
System.NullReferenceException: Object reference not set to an instance of an object.
   at Carbon.Scheme.SchemeUtils.get_EditorMode()
   at Carbon.Web.Plugins.PlgSchBasicCompSpec.Init()
   at Carbon.Web.AppData.InitPlugins()

closed 2018-04-21 by 2mik