RapidScada/scada #57: VS Project File conflict

lugreen, 2018-04-21:
[image]

circler3, 2018-04-21:
Encodings can be replaced by visual studio with different culture settings. Keeping the original one is OK. It won't trouble your visual studio.

FYI, please provide some description about the issue when you are going to start a issue. More information about the project can be found at the [forum](http://forum.rapidscada.org).

lugreen, 2018-04-21:
Hi,
Thanks very much.
Ths VS will auto change the encoding, it will case some touble when others developer fork the project and commit code.
BTW,I find that the project file(.csproj) 'encoding using utf-8 will not be changed by VS.

circler3, 2018-04-21:
Hi,
If you are going to contribute to the repo, please make PRs to `develop` branch. And make sure you are in compliance with the [roadmap](https://forum.rapidscada.org/?topic=roadmap).

2mik, 2018-04-21:
Hi,
It's better to use utf-8. Using windows-1251 is legacy. 
@circler3 is right, don't commit to master, commit to develop instead. Before development new features, we could discuss them on https://forum.rapidscada.org/
Also I plan to create a repository "scada-communitry" for commits that don't affect the core of Rapid SCADA. I can do it anytime when it's needed to somebody.