RapidScada/scada #60: Siemens S7 support by Sharp7

Dunkelicht, 2018-05-30:
Greetings, thanks a lot for creating this project, it seems stable.
Siemens PLC is popular , please consider support s7 protocol by using sharp7. It is also open source. 
http://snap7.sourceforge.net/sharp7.html 

Good luck.

2mik, 2018-05-31:
Hi,
To use snap7, it should be wrapped in a new Rapid SCADA driver. If you are a developer, you could do this.
Also you can connect to S7 by Modbus, see this [link](https://support.industry.siemens.com/cs/document/22660304/how-do-you-establish-a-modbus-tcp-communication-from-a-simatic-s7-300-s7-400-and-where-can-you-find-further-information-?dti=0&lc=en-WW)

2mik, 2018-05-31:
Thank you for the information about sharp7

Dunkelicht, 2018-06-01:
Thanks for the info, I will try to do so and update it when I finish it.

Sharp7 is GPL, maybe it is better use snap7.dll so that it will not change RapidSacad license to GPL.

2mik, 2018-06-01:
Most likely, the licenses are compatible, see [this link](https://opensource.stackexchange.com/questions/5664/linking-from-lgpl-2-1-software-to-apache-2-0-library). You can ask the developers of sharp7 about compatibility.

2mik, 2018-06-01:
From [here](https://www.apache.org/licenses/GPL-compatibility.html):
_This licensing incompatibility applies only when some Apache project software becomes a derivative work of some GPLv3 software, because then the Apache software would have to be distributed under GPLv3._

2mik, 2018-06-01:
A note: sharp7 is not GPL, it is LGPL. They are different.

2mik, 2018-06-01:
[Example](https://github.com/rubychan/coderay/issues/25) of conversation about licenses.

circler3, 2018-06-01:
FYI, for PLC access, [HslCommunication](https://github.com/dathlin/HslCommunication) is a good option. It is capable of  melsec plc, siemens plc and omron plc. License is LGPLv3. Linking LGPLv3 libraries will not change the original license. Although a license declaration is required.

closed 2018-06-02 by 2mik