RapidScada/scada #61: Communicator - Dynamic  Interaction between KPLogic and KPView

mhanuel26, 2018-06-30:
Hello Mikhail,

I am trying to do some dynamic menu for Communicator and so far I haven't able to do it, I am not positive if I am missing something that might help me in the task.

I have an Ethernet to Serial device that has a communication TCP port (different from Data) for configuration, the modbus KLogic using TCP client as communicator channel works really good for the data channel.

I want to integrate the configuration under Rapid Scada, so far I have the following basic Forms created and working.

[image]

The Serial Menu opens after the first one using the Serial Settings button. 
What I want is to be able to send configuration data using a button, such as the Apply button without leaving the UI. The problem is that I cannot use Connection property from KPView.

I have tried the following as a possible workaround without success

The example follows the KpTestLogic code (actually I did my project under this)

I defined a Connection property 
`private static Connection baseConnection;`
Then I assign it's value under the OnConnectionSetmethod such as 

```
        /// <summary>
        /// Perform actions after the connection is established
        /// </summary>
        public override void OnConnectionSet()
        {
            baseConnection = Connection;
            FileStream fileStream = new FileStream("C:\\miLog.txt", FileMode.Create);
            StreamWriter sw = new StreamWriter(fileStream);
            sw.WriteLine("(Constructor) baseConnection = " + baseConnection.ToString());
            sw.Close();
        }
```
The ugly part is to have a static method that do the write

```
        public static bool SendCommandToWiznet()
        {
            if (baseConnection == null || !baseConnection.Connected)
            {
                FileStream fileStream = new FileStream("C:\\miLog.txt", FileMode.Append);
                StreamWriter sw = new StreamWriter(fileStream);
                sw.WriteLine("baseConnection NULL or Not Connected");
                sw.Close();
                return false;
            }
            else
            {
                baseConnection.WriteLine("Test");
                return true;
            }
        }
```
from the Apply Button method I call it 

```
        private void btnApply_Click(object sender, EventArgs e)
        {
            KpTestLogic.SendCommandToWiznet();
        }
```

It seems the _baseConnection_ property is always null.

But from Session I can call the _SendCommandToWiznet_ method and it works.

The reason I want to have it call from UI is because I have some GPIO that I want to visualize dynamically there, but also because I found the Session the only place to send over the channel and it seems it's called periodically based on the period defined in main window, which will make the settings taking effect dependable on that period.

I have study even the Communicator code without success since I don't know how can I get a reference to Connection in KPView, but also I understand it might not be a good idea to modify upper code, unless it take the forms of callbacks such asn OnConnectionXXX sort of. 

For example a method that get called after closing the windows might help, such as OnPropertyViewClose...

I will appreciate your response,

2mik, 2018-06-30:
Hi,

Instance of KPLogic works as a part of ScadaCommSvc.exe while instance of KPView is created by ScadaCommCtrl.exe. So you can't access from UI to an instance of Connection that is in the different application.
You can interact by using read/write to files, WCF service or other method of communication between applications. You can also read the configuration ScadaCommSvcConfig.xml by the existing class if needed.

mhanuel26, 2018-06-30:
Hi Mikhail,

Thank you for your response,

2mik, 2018-07-02:
Be free to ask.

closed 2018-07-02 by 2mik