RapidScada/scada #62: Error on opening Log.std project in in Visiual Studio 2015

giplg, 2018-07-03:
Loading of Log.std project in VS 2015 failed with error:
d:\proj\RapidScada\Log\Log.Std\Log.Std.csproj : error  : The default XML namespace of the project must be the MSBuild XML namespace. If the project is authored in the MSBuild 2003 format, please add xmlns="http://schemas.microsoft.com/developer/msbuild/2003" to the <Project> element. If the project has been authored in the old 1.0 or 1.2 format, please convert it to MSBuild 2003 format.  d:\proj\RapidScada\Log\Log.Std\Log.Std.csproj`
Comments in Google say it's because of project was opened/saved in Visual Studio 2017, having new project file format.  Same error in ScadaData project.

2mik, 2018-07-04:
You should use VS 2017.

giplg, 2018-07-04:
Ok, than „HowToBuild.txt” should be corrected – it says about VS 2015:
[cid:image002.png@01D413A2.3ECE3C80]

2mik, 2018-07-06:
You right. Already corrected in the develop branch.
Thank you!

closed 2018-07-06 by 2mik