RapidScada/scada #75: Possible XSS vulnerabilities due to Bootstrap/jQuery

KaiNahrgang, 2019-03-27:
Hi,
the latest version of Rapid Scada still uses jQuery 2.2.3 and Bootstrap 3.3.6, which both have security flaws. Thus, I recommend you to update the libraries to their latest version.

Bootstrap: 
- CVE-2019-8331 (fix version: 3.4.1)
- CVE-2018-14041 (fix version: 4.1.2)
- CVE-2018-14040 (fix version: 4.1.2)
- CVE-2018-14042 (fix version: 4.1.2)

jQuery:
- CVE-2015-9251 (fix version: 3.0.0)

Best regards

2mik, 2019-03-28:
Hi,
Thank you for the suggestion.
Do you have any cases how to hack Rapid SCADA?

KaiNahrgang, 2019-03-28:
Hi,
to be honest, I haven't looked into it yet, so there is a chance that those are only theoretical flaws.

2mik, 2019-03-28:
We will update the versions of the libraries on the next iteration of development of the web app.

2mik, 2021-11-15:
Bootstrap will be updated in Rapid SCADA 6.

closed 2021-11-15 by 2mik